jeffeb3/sandify #237: Add new "code" layer

michaelcunningham1974, 2022-08-09:
It would be useful to have a new layer type called code that would provide a textbox that arbitrary code could be entered similar to the start/end code in the machine settings.

I envision this being useful to do things like pauses  between layers, changing federates, changing led settings, custom pin/io etc.

jeffeb3, 2022-10-19:
This is several months old now, I'm not sure how I missed it. 

That would be easy enough to add. My concern is just being able to get the user to understand it. We wouldn't be able to have any kind of preview for that text, it would just be a layer that gets copy/pasted at the end. I'm not 100% sure our architecture can manage that kind of storage through to the download ATM either.

bobnik, 2024-11-03:
Added in #302

closed 2024-11-03 by bobnik