jeffeb3/sandify #239: Local install OSX

cronos45, 2022-09-22:
Hello, I have tried to install this for local use but am encountering issues. I have included my log.

[2022-09-22T09_00_29_417Z-debug-0.log.zip](https://github.com/jeffeb3/sandify/files/9624248/2022-09-22T09_00_29_417Z-debug-0.log.zip)

jeffeb3, 2022-09-22:
```
10918 info run node-sass@5.0.0 postinstall { code: 1, signal: null }
10919 timing reify:rollback:createSparse Completed in 5913ms
10920 timing reify:rollback:retireShallow Completed in 0ms
10921 timing command:install Completed in 219578ms
10922 verbose stack Error: command failed
10922 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:63:27)
10922 verbose stack     at ChildProcess.emit (node:events:513:28)
10922 verbose stack     at maybeClose (node:internal/child_process:1093:16)
10922 verbose stack     at Socket.<anonymous> (node:internal/child_process:451:11)
10922 verbose stack     at Socket.emit (node:events:513:28)
10922 verbose stack     at Pipe.<anonymous> (node:net:757:14)
10923 verbose pkgid node-sass@5.0.0
10924 verbose cwd /Users/anthonycruz/Desktop/sandify-0.2.6
10925 verbose Darwin 20.6.0
10926 verbose node v16.17.0
10927 verbose npm  v8.15.0
10928 error code 1
10929 error path /Users/anthonycruz/Desktop/sandify-0.2.6/node_modules/node-sass
10930 error command failed
10931 error command sh /var/folders/gx/c_8cmh4905g898c3vq1q51480000gn/T/postinstall-4e29d143.sh
10932 error Building: /usr/local/bin/node /Users/anthonycruz/Desktop/sandify-0.2.6/node_modules/node-gyp/bin/node-gyp.js rebuild --verbose --libsass_ext= --libsass_cflags= --libsass_ldflags= --libsass_library=
10932 error   c++ -o Release/obj.target/libsass/src/libsass/src/ast.o ../src/libsass/src/ast.cpp '-DNODE_GYP_MODULE_NAME=libsass' '-DUSING_UV_SHARED=1' '-DUSING_V8_SHARED=1' '-DV8_DEPRECATION_WARNINGS=1' '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D_GLIBCXX_USE_CXX11_ABI=1' '-D_DARWIN_USE_64_BIT_INODE=1' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DLIBSASS_VERSION="3.5.5"' -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/include/node -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/src -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/openssl/config -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/openssl/openssl/include -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/uv/include -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/zlib -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/v8/include -I../src/libsass/include  -O3 -gdwarf-2 -mmacosx-version-min=10.7 -arch x86_64 -Wall -Wendif-labels -W -Wno-unused-parameter -std=c++11 -stdlib=libc++ -fno-strict-aliasing -MMD -MF ./Release/.deps/Release/obj.target/libsass/src/libsass/src/ast.o.d.raw   -c
10932 error   c++ -o Release/obj.target/libsass/src/libsass/src/ast_fwd_decl.o ../src/libsass/src/ast_fwd_decl.cpp '-DNODE_GYP_MODULE_NAME=libsass' '-DUSING_UV_SHARED=1' '-DUSING_V8_SHARED=1' '-DV8_DEPRECATION_WARNINGS=1' '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D_GLIBCXX_USE_CXX11_ABI=1' '-D_DARWIN_USE_64_BIT_INODE=1' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DLIBSASS_VERSION="3.5.5"' -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/include/node -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/src -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/openssl/config -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/openssl/openssl/include -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/uv/include -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/zlib -I/Users/anthonycruz/Library/Caches/node-gyp/16.17.0/deps/v8/include -I../src/libsass/include  -O3 -gdwarf-2 -mmacosx-version-min=10.7 -arch x86_64 -Wall -Wendif-labels -W -Wno-unused-parameter -std=c++11 -stdlib=libc++ -fno-strict-aliasing -MMD -MF ./Release/.deps/Release/obj.target/libsass/src/libsass/src/ast_fwd_decl.o.d.raw   -c
...
10933 error gyp info it worked if it ends with ok
10933 error gyp verb cli [
10933 error gyp verb cli   '/usr/local/bin/node',
10933 error gyp verb cli   '/Users/anthonycruz/Desktop/sandify-0.2.6/node_modules/node-gyp/bin/node-gyp.js',
10933 error gyp verb cli   'rebuild',
10933 error gyp verb cli   '--verbose',
10933 error gyp verb cli   '--libsass_ext=',
10933 error gyp verb cli   '--libsass_cflags=',
10933 error gyp verb cli   '--libsass_ldflags=',
10933 error gyp verb cli   '--libsass_library='
10933 error gyp verb cli ]
10933 error gyp info using node-gyp@7.1.2
10933 error gyp info using node@16.17.0 | darwin | x64
10933 error gyp verb command rebuild []
10933 error gyp verb command clean []
10933 error gyp verb clean removing "build" directory
10933 error gyp verb command configure []
10933 error gyp verb find Python Python is not set from command line or npm configuration
10933 error gyp verb find Python Python is not set from environment variable PYTHON
10933 error gyp verb find Python checking if "python3" can be used
10933 error gyp verb find Python - executing "python3" to get executable path
10933 error gyp verb find Python - executable path is "/Library/Frameworks/Python.framework/Versions/3.9/bin/python3"
10933 error gyp verb find Python - executing "/Library/Frameworks/Python.framework/Versions/3.9/bin/python3" to get version
10933 error gyp verb find Python - version is "3.9.4"
10933 error gyp info find Python using Python version 3.9.4 found at "/Library/Frameworks/Python.framework/Versions/3.9/bin/python3"
10933 error gyp verb get node dir no --target version specified, falling back to host node version: 16.17.0
10933 error gyp verb command install [ '16.17.0' ]
[... cut ...]
10933 error gyp ERR! command "/usr/local/bin/node" "/Users/anthonycruz/Desktop/sandify-0.2.6/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--verbose" "--libsass_ext=" "--libsass_cflags=" "--libsass_ldflags=" "--libsass_library="
10933 error gyp ERR! cwd /Users/anthonycruz/Desktop/sandify-0.2.6/node_modules/node-sass
10933 error gyp ERR! node -v v16.17.0
10933 error gyp ERR! node-gyp -v v7.1.2
10933 error gyp ERR! not ok 
10933 error Build failed with error code: 1
10934 verbose exit 1
```
node-sass is trying to build C++ code on your machine, and it has a C++ error. Is this a vanilla checkout of sandify?

I will try doing a fresh clone on my Linux machine.

cronos45, 2022-09-22:
@jeffeb3 yes this was a vanilla install, I can try removing it and reinstalling but would need some guidance to do so.

jeffeb3, 2022-09-22:
I don't have a mac. @bobnik does, I think. 

If you remove the node_modules folder and run npm install again, it will do a clean build.

cronos45, 2022-09-22:
This was the result after deleting the node within the sandy folder.
[2022-09-22T17_46_41_765Z-debug-0.log.zip](https://github.com/jeffeb3/sandify/files/9628296/2022-09-22T17_46_41_765Z-debug-0.log.zip)

bobnik, 2022-09-22:
@cronos45 @jeffeb3 I do have a Mac, and I will investigate, but it may be a few days. I'll circle back.

bobnik, 2022-11-06:
@cronos45, sorry for the delayed response. This issue was due to an incompatible (older) version of node-sass, which has since been upgrade. Please pull the latest when you can, and try to rebuild, thanks.

closed 2023-10-14 by bobnik