jeffeb3/sandify #256: Starting GCode getting reformatted

V1EngineeringInc, 2022-11-26:
When I enter this starting GCode it always gets reformatted, in a way that the machine will not run properly. I have to manually edit all my files before running them.

`; BEGIN PRE
$HY
G0 Y-4 F50
G92 Y0
$HX
G0 X-13 F50
G92 X0
G1 X1 F1000
; END PRE`

That is what I need,

`; BEGIN PRE
$HY G0 Y-4 F50
G92 Y0
$HX
G0 X-13 F50
G92 X0
G1 X1 F1000
; END PRE`

That is what I get. I have to edit that first line. Any ideas? I made sure I did not have any extra spaces or anything.  The second $HX command seems to work fine so not a "$" thing?

V1EngineeringInc, 2022-11-26:
Had to add line breaks to get it to look okay. It comes out with no spaces, as expected from Sandify.

jeffeb3, 2022-11-26:
I was worried it was the `$HY`, because I didn't think those were allowed in gcode. But it isn't that, it is the first line always getting joined with the second:

```
One
Two
Three
```
Becomes:
```
; BEGIN PRE
One Two
Three
; END PRE
```

Thanks for reporting it. It should be an easy fix.

V1EngineeringInc, 2022-11-26:
Well, bummer it is a bug, but glad I wasn't just doing something wrong.

I have a fun gcode running right now, can't wait to show you....and it lead to an idea.

jeffeb3, 2022-11-26:
If you just add an empty line as the first line, it will be fine. 

Bob and I are working on a bunch of changes at once in a restructuring to enable some new features. IDK when that will be finished. I'm interested to see what you've come up with, and make sure it fits in the new system.

V1EngineeringInc, 2022-11-26:
Nice tip I will do that.

bobnik, 2024-10-24:
Fixed in #302

closed 2024-10-24 by bobnik