jeffeb3/sandify #258: Export Adds Points

jeffeb3, 2022-11-29:
https://forum.v1engineering.com/t/sandify-update-v0-2-7/34684/21?u=jeffeb3

[coquillage (1).zip](https://github.com/jeffeb3/sandify/files/10114483/coquillage.1.zip)

[image]

I don't think it has to do with the "connect to next layer" option. It seems to travel from the end of pattern 1 to the beginning of pattern 3, then back to the beginning of pattern 2. I have to have at least the first three imports enabled to see it.

V1EngineeringInc, 2022-11-29:
This happened on my logo pattern. It added exactly one extra point. I found it odd. I thought I had maybe added a point layer, but that would have added two point, start and end, but it didn't (I don't think).

bobnik, 2023-10-13:
#275 may have fixed this. Please try again, thanks.

bobnik, 2023-10-14:
I'm going to go ahead and close this. Please submit another issue with a .sdf file as example if you see it again.

closed 2023-10-14 by bobnik