jeffeb3/sandify #265: G-code target exceeds machine limits error in octoprint.

jacobmcclellan, 2023-05-02:
Hello, I have noticed that when I make a drawing that 'exceeds' the dimensions of the drawing space in Sandify although it shows that the lines draw to the edge and then move along the edge to the place where it would enter back into the drawing space Octoprint still says that it exceeds the bounds of the machine. If I move the shape being drawn so it doesn't touch or pass the edge, it will work as expected. I have set the limits on Sandify to be slightly smaller than my machine to ensure I didn't accidentally type in the wrong numbers. There was a setting that I turned on to not exceed the edge and it did not resolve the issue. Is this an issue with Sandify or somewhere else? I am new to Sandify and Octoprint (I just finished building a custom sand drawing table) so I could totally be doing something wrong and Sandify could be working fine. If you would like I can post links of pastbin-ed gcode if there would be issues that could be seen that way.

bobnik, 2023-10-14:
Please use the current 1.0.0 version and upload a .sdf (save) file that illustrates this issue, and we'll take a look.

bobnik, 2024-10-12:
Closing due to inactivity...

closed 2024-10-12 by bobnik