jeffeb3/sandify #276: Preview render missing when width is 992px

aaronse, 2023-10-14:
Neat project, nice update!   Noticed preview render is not displayed when width is 992px.  Consider checking out responsive layout breakpoint logic.  Cheers!

bobnik, 2023-10-14:
Yes, it's all built around responsive breakpoint logic. Seems we're off by a pixel point though lol. Only 992px exactly causes this issue - good catch. We'll fix.

aaronse, 2023-10-14:
Only noticed because I like how your project is looking as was playing around to see what your responsive layouts look like :-) 
Thoughts on https://github.com/jeffeb3/sandify/pull/277 ?

aaronse, 2023-10-14:
Built and verified fix... 

[image]

bobnik, 2023-10-15:
Fixed by #277

closed 2023-10-15 by bobnik