jeffeb3/sandify #278: "npm start" failing with "JSX syntax extension is not currently enabled"

aaronse, 2023-10-14:
Am following https://github.com/jeffeb3/sandify#running-from-source-locally

"npm start" failing with "JSX syntax extension is not currently enabled.

Error message seems to include info on how to fix.  Am new to vite, so figured on filing this issue for clarity...

"The esbuild loader for this file is currently set to "js" but it must be set to "jsx" to be able to parse JSX syntax. You can use "loader: { '.js': 'jsx' }" to do that."

aaronse, 2023-10-14:
Hmm, maybe those messages are ignorable?  Despite the scary errors during startup am still able to access the localhost:3000 app 🤷‍♂️ ?

[image]

bobnik, 2023-10-14:
Yes, we have configuration settings in vite.config.js to handle that error, so not sure why are you seeing it.

What version of node are you using? We are at 20.6.1. 

Are you using nvm for node versioning? If not, I recommend installing it, installing 20.6.1 if you don't have it, and then running "nvm use" from the project directory. Likely unrelated, but I see no warnings when running "npm start".

aaronse, 2023-10-14:
Am on Node 18 still.  Site still seems to work, will update to 20 if I run into a blocking issue, and/or when Node 20 reaches long term support https://nodejs.org/en/blog/announcements/v20-release-announce .   Cheers for the info/guidance!

closed 2023-10-14 by aaronse

jeffeb3, 2023-10-14:
I use the docker compose. There important commands are in the docker compose file. So that should always have a working version of node and npm.