jeffeb3/sandify #313: filter using shapes like star, clover etc.

jattie-ire, 2025-02-02:
It's tedious and prone to errors to creates filters using circle and rectangle only, can filters include all the existing standard shapes like star, polygon etc?

jeffeb3, 2025-02-06:
The masks we have are basically a copy of the rectangle and circular limits of the machines. The limit code is surprisingly complex and brittle. Making it work for an arbitrary polygon (especially ones with curves like the releaux triangle, concave angles, or self intersections) would be pretty hard. The limits and masks are also pretty computationally intensive, since they have to work for every line segment. So if you have a complex mask, the line segment intersections would be pretty difficult.

I'm not saying it's impossible. But it is more complex than it seems. It would be more likely that a triangle or star would be a hard coded mask, one at a time. We would need some clever geometry to make it simpler and it would depend on the shape.

I would love to see any of the wipers constrained by the star though. That would be cool.

jattie-ire, 2025-02-06:
Thanks, I understand,I was actually trying to mask of a wiper  unsuccessfully using multiple squares and it gets quite complicated fast.